mathertel/DmxSerial2 #10: Device doesn't unmute when the unmute request is sent to broadcast uid

rbarreiros, 2018-04-09:
As the title states, it doesn't unmute when the request is made to broadcast as it only checks if packet is to self, should check also if it's to all.

I won't push a patch as the fix is simple enough.

DmxSerial2.cpp, line 525 should be:
 if (packetIsForMe || packetIsForAll) { 

Best regards,

peternewman, 2018-04-09:
It's actually going to want to be
`if (packetIsForMe || packetIsForGroup || packetIsForAll) {`

Likewise line 541 for the mute.

How did you find out @rbarreiros ; I see OLA doesn't currently have tests for this:
https://github.com/OpenLightingProject/ola/blob/master/tools/rdm/TestDefinitions.py

rbarreiros, 2018-04-09:
Hey Peter,
Hehehe, forgot the packetIsForGroup also. I thought the mute would respond to self only ? 
I found it while using an arduino loaded with the responder example of DmxSerial2 as a test for an artnet controller with RDM support, initially with an STM32F103 (blue pill) but it's getting hard to stick everything in 64k flash and 20k ram....

peternewman, 2018-04-09:
Yeah I wasn't sure either, but the standard says:

> 7.6.3 Discovery Mute Message (DISC_MUTE)
> A responder port shall set its Mute flag when it receives this message containing its UID, or a broadcast address.

Which is about as clear as you get for the RDM spec :smile: , I'm taking plural of broadcast address to include vendorcast too.

peternewman, 2018-08-19:
It is actually a bit more subtle, as we only want to respond to the unicast, but action any of uni, vendor or broadcast.
PR here, but needs testing:
https://github.com/mathertel/DmxSerial2/pull/13

@rbarreiros do you still have your artnet controller?

rbarreiros, 2018-08-21:
Hey Peter, yes I do, but it's development is in standby, festival season, currently on small vacation, hope to get back at it late september, will test as soon as possible.

peternewman, 2018-08-21:
That's cool, I'm busy for a bit too. I'm hoping to do some testing on an Arduino base too, and ideally sort some OLA tests too.

PR #13 (2018-08-22): Untested fix to ensure we act on vendorcast/broadcast mute/unmute
Untested, but should close #10
commit d546ea54 (2018-08-22): Untested fix to ensure we act on vendorcast/broadcast mute/unmute

Untested, but should close #10
closed 2018-08-22 by mathertel