mathertel/DmxSerial2 #11: Example using sensors

kidpatchcabbage, 2018-05-13:
Please create an example using a sensor. I am a noob and struggling to implement.
Any help appreciated. Thanks.

peternewman, 2018-08-18:
Well @StefanKrupop added the sensor code in https://github.com/mathertel/DmxSerial2/pull/7 . Do you have an example Stefan?

jsphpl, 2020-05-13:
@StefanKrupop posted an example using sensors over there: https://forum.dmxcontrol-projects.org/index.php?thread/13724-adding-sensors-to-arduino-rdm-dmx/&postID=115681#post115681

peternewman, 2021-06-08:
@StefanKrupop I think it would be great to get your example merged . I'm happy to work on it and perhaps switch the sensors to the onboard temperature and voltage ones so they'll just work for everyone out of the box.

Do you want to take credit for your work? In which case could you push it up to a fork of this repo and I can make changes on top of that, or are you happy for me to commit something inspired by your code?

StefanKrupop, 2021-06-09:
Feel free to take my example sketch (which really isn't much) and make a nicer example out of it. ;-)