mathertel/DmxSerial2 #15: Verify error

claysam75, 2018-09-15:
Verifying in Arduino 1.8.5 returns;

RDMSerialRecv:67: error: too many initializers for 'RDMINIT'
 };

peternewman, 2018-09-16:
Was this on the code downloaded from GitHub @claysam75 ? The number of arguments seems right to me by observation, but I've not tried a verify, so there may be something we're doing that's upsetting it.

peternewman, 2018-09-16:
I get a different error on 1.8.2:
```
/tmp/arduino_modified_sketch_43130/RDMSerialRecv.ino:67:1: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
 };
 ^
/tmp/arduino_modified_sketch_43130/RDMSerialRecv.ino:67:1: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
```

This PR fixes that (I don't know if it fixes your issue too?):
https://github.com/mathertel/DmxSerial2/pull/16

claysam75, 2018-09-17:
Hi, I applied your fix but the issue persists. Now running 1.8.7

peternewman, 2018-09-18:
Hmm, is there anything else logged, or just that one line?

claysam75, 2018-09-18:
Screenshot of error output is attached (with just my google drive email covered)

[image]

peternewman, 2018-09-26:
@mathertel this needs re-opening as #15 doesn't actually fix it unfortunately.

claysam75, 2018-10-04:
@mathertel Any update on this error?

peternewman, 2018-10-27:
Bump @mathertel , can you reopen this please?

PR #16 (2018-09-26): Change the static strings to const char*
Hopefully closes #15 

Untested on hardware
commit 22cd830b (2018-09-26): Change the static strings to const char*

Hopefully closes #15 

Untested on hardware