mathertel/DmxSerial2 #20: Build warnings (part 3)

g0uus, 2018-10-15:
If the Arduino Compiler Warnings option is set to YES, the following warnings are generated -
```
C:\Users\grh\Google Drive\Arduino\libraries\DmxSerial2\src\DMXSerial2.cpp: In member function 'void DMXSerialClass2::write(int, uint8_t)':

C:\Users\grh\Google Drive\Arduino\libraries\DmxSerial2\src\DMXSerial2.cpp:411:13: warning: comparison is always false due to limited range of data type [-Wtype-limits]

   if (value < DMXSERIAL_MIN_SLOT_VALUE) value = DMXSERIAL_MIN_SLOT_VALUE;

             ^

C:\Users\grh\Google Drive\Arduino\libraries\DmxSerial2\src\DMXSerial2.cpp:412:13: warning: comparison is always false due to limited range of data type [-Wtype-limits]

   if (value > DMXSERIAL_MAX_SLOT_VALUE) value = DMXSERIAL_MAX_SLOT_VALUE;

             ^
```

The problem here is that `value` is declared as `uint8_t` and can thus never be less than `DMXSERIAL_MIN_SLOT_VALUE` or greater than `DMXSERIAL_MAX_SLOT_VALUE`

The fix is to simply remove or comment out the two if statements.

peternewman, 2018-10-16:
See the discussion here https://github.com/mathertel/DmxSerial2/issues/18#issuecomment-430237661

In this case you probably want to add a comment saying what the old test was and maybe why it's been removed.

mathertel, 2018-10-27:
I commented out with a comment.
https://github.com/mathertel/DmxSerial2/commit/66ed0cffcb398b8912e510eff7ae6efeff62503a

No effect on code size. I assume the compiler already removed them during optimization.

Thanks.

closed 2018-10-27 by mathertel