mathertel/DmxSerial2 #28: RDM channel size from 3 to 4

jabadabado, 2020-05-18:
Anyone knowing how to increase the number of channels from 3 to 4. The example have 3 channels, RGB for LED's, I would like to add an additional channel. I can't find where to change 3 to 4, anyone who knows where to change it?

peternewman, 2020-05-18:
Hi @jabadabado ,

Setting the 3 to a 4 here will change the number of channels it says it needs via RDM:
https://github.com/mathertel/DmxSerial2/blob/f4722dfc8b32265d36ad64603a34f520245a6465/examples/RDMSerialRecv/RDMSerialRecv.ino#L66

From a PWM output point of view, it's being done here:
https://github.com/mathertel/DmxSerial2/blob/f4722dfc8b32265d36ad64603a34f520245a6465/examples/RDMSerialRecv/RDMSerialRecv.ino#L43-L58
And:
https://github.com/mathertel/DmxSerial2/blob/f4722dfc8b32265d36ad64603a34f520245a6465/examples/RDMSerialRecv/RDMSerialRecv.ino#L93-L102
And
https://github.com/mathertel/DmxSerial2/blob/f4722dfc8b32265d36ad64603a34f520245a6465/examples/RDMSerialRecv/RDMSerialRecv.ino#L132-L151

wad77, 2024-01-18:
Hello, I'm a complete newbie to Arduino. Please write how to make 6 PWM channels on Arduino UNO. I beg you very much. I don't know how to edit code. THANK YOU VERY MUCH!

mathertel, 2024-01-19:
> Hello, I'm a complete newbie to Arduino. Please write how to make 6 PWM channels on Arduino UNO. I beg you very much. I don't know how to edit code. THANK YOU VERY MUCH!

The best you can do is learning programming and take the ownership of what you need and have done.