mathertel/DmxSerial2 #31: Add support for STM32

sixstringsg, 2021-04-26:
How possible would it be to add support for the STM32 Arduino core? Right now there's a whole pile of compile errors similar to `'UCSRnB' was not declared in this scope`

peternewman, 2021-04-26:
@midikidi has apparently done it already, see some of the discussion in https://github.com/mathertel/DmxSerial2/issues/29 .

I suspect it could be fixed with https://www.deviceplus.com/arduino/arduino-preprocessor-directives-tutorial/ or similar.

Have you tried setting this:
https://github.com/mathertel/DmxSerial2/blob/853782165054acb0b4e17b045c35b05818a8742c/src/DMXSerial2.cpp#L49-L53

Which might declare that value.

midikidi, 2021-04-26:
Hi Sixstringsg, As Peter said I have ported it to the STM32 but It is built with the STM32 Cube IDE.  I guess it would be fairly straight forward to port to the Arduino IDE. IF Arduino knows about STM Uart registers and supports HAL functions. I have implemented 12 timers to give me hi resolution PWM all RDM is working I am working on 6 personalities almost there  I can select them but cant get my the personality description working.

IS512, 2021-04-26:
ESP32 Arduino Core support too would be handy ;)

midikidi, 2021-04-26:
Done a lot of work with DMX and had a lot of issues with ESP32 loosing DXM data the ESP8266 is better. I think the ESP32 with all its background tasks /2 core and the Uart itself is a place to start. Even Claudeheintz a wiz with DMX says the same. Sparkfun has code for there ESP32 shield and that looses data in the higher DMX addresses. Sparkfun doesn't do any thing fancy to detect break but the claudeheintz library does.

IS512, 2021-04-26:
ESP32 can do high-frequency 16bit PWM, Yeah Claudes library has some issues to sort out. 

This library has been great so far but obviously lacks the complexity of RDM: 
[https://github.com/luksal/ESP32-DMX-RX](url)

sixstringsg, 2021-04-26:
@midikidi would you mind sharing your copy of the library with STM32 support? Or is the zip you posted in your issue the latest? I would love to try and get it to work in the Arduino IDE, [The STM32 Arduino Core](https://github.com/stm32duino/Arduino_Core_STM32) appears to support HAL & UART.

midikidi, 2021-04-27:
Happy to forward the dmx part as there's bits I cant share, but it should help. if you manage to get personalites working it would help me. Im a novice with github so no sure how to send zip

midikidi, 2021-04-27:
ok sussed it
[temp.zip](https://github.com/mathertel/DmxSerial2/files/6381062/temp.zip)

peternewman, 2021-04-27:
@midikidi my comments in https://github.com/mathertel/DmxSerial2/issues/29#issuecomment-814944905 still stand. If you put that code in a repo somewhere (ideally a fork of this repo) I'm happy to review and comment on the personality stuff and can probably point you in the right direction to get it working.

midikidi, 2021-04-27:
Sixstringsg feel free to use the code as you like its for anyone Appreciate you asking  Im up against a deadline on a project and no time to get to grips with github busy coding (LEDSynth) c++/c/html/js/css/arduino and stm cube .Peter any help with Sixstrinsg's efforts greatly appreciated

arielreyes2021, 2022-03-30:
Hello, I wanted to know if it was possible to get a dmx library for stm32... I have a board with a stm32f407ve and I would need a dmx library...

riwalker, 2023-02-09:
Will this code run on any STM32 Micro ?
any specific requirements ? i.e 2 UART, >32K Flash etc ?
and will this be ported if ifdef comments to the standard DMXSERIAL library ?

madfishy, 2024-10-15:
Hello There,
Did support for STM32 ever get added to DmxSerial2? or does someone have an Arduino sketch with @midikidi running that they could share.

Thanks in advance
Regards
E