mathertel/DmxSerial2 #32: DMX Out

Kevitto, 2021-05-19:
Good morning!

I've built an RDM IN module with this code, which works awesome, by the way!  But now, I'm trying to get it to ingest RDM through the serial port and push out DMX at address 1 from the received packets, but for some reason using this in tandem with DMXSimple kills the DMXSerial2.tick() timer (it'll only respond to RDM once, then stop responding) when I try to use the DMX out.

Is there a way to implement DMX forwarding directly in this code?  Not all the functions are exposed in the Wiki and documentation.

Thanks!

peternewman, 2021-05-20:
Hi @Kevitto ,

If I understand correctly, you're essentially trying to build one of these?:
http://www.farnell.com/datasheets/1655633.pdf

So having the DMXSimple code present, but not calling DMXSimple.write() doesn't break it for example?

Can you add some form of debugging to see what's actually breaking (e.g. does it still receive new DMXSerial2 DMX data into the unit?

Sharing the code would be great in general and might help to debug this. IMHO it would make an excellent example piece of code as it might even be a fairly common thing to want to make.

Kevitto, 2021-05-25:
Yes, I was able to fix the code so DMX out throws to DMX address 1 regardless of RDM address and only to the pre-programmed size of the personality I set up.  It also has a built-in function for identify.  Here is the code!

I'm wondering how to remove all the unused options in the RDM menu of the device, so I only show the basics and not the special personalities the unit comes pre-programmed with.

`// - - - - -
// DmxSerial2 - A hardware supported interface to DMX and RDM.
// RDMSerialRecv.ino: Sample RDM application.
//
// Copyright (c) 2011-2013 by Matthias Hertel, http://www.mathertel.de
// This work is licensed under a BSD style license. See http://www.mathertel.de/License.aspx
//
// The following RDM commands are implemented here:
// E120_LAMP_HOURS
// E120_DEVICE_HOURS
//
// More documentation and samples are available at http://www.mathertel.de/Arduino
// - - - - -

#include <DmxSimple.h>
#include <EEPROM.h>
#include <DMXSerial2.h>

// see DMXSerial2.h for the definition of the fields of this structure
//const uint16_t my_pids[] = {E120_DEVICE_HOURS, E120_LAMP_HOURS};
const uint16_t my_pids[] = {};
struct RDMINIT rdmInit = {
  "Company Name", // Manufacturer Label
  1, // Device Model ID
  "RDM Module", // Device Model Label
  3, // footprint
  (sizeof(my_pids)/sizeof(uint16_t)), my_pids
};

uint16_t startAddress = 0;
int maxSize = 3;

void setup () {
  DMXSerial2.init(&rdmInit, processCommand);
  
  pinMode(9, OUTPUT);
  digitalWrite(9, LOW);
  DmxSimple.usePin(3);
  DmxSimple.maxChannel(3);

} // setup()

void loop() {
    if (DMXSerial2.isIdentifyMode()) {
      
      digitalWrite(9, HIGH); // indicator light for the Identify mode
      
    } else {
      startAddress = DMXSerial2.getStartAddress(); // this should work... but it doesn't?
      
      for (int i = 0; i < maxSize; i++){ // for all DMX packets sent to addresses up to maxSize, forward to DMX out
        
        DmxSimple.write(i + 1, DMXSerial2.readRelative(i));
      }
      
      digitalWrite(9, LOW);
    }
    
  DMXSerial2.tick();
} // loop()

// This function was registered to the DMXSerial2 library in the initRDM call.
// Here device specific RDM Commands are implemented.
bool8 processCommand(struct RDMDATA *rdm, uint16_t *nackReason)
{
  byte CmdClass       = rdm->CmdClass;     
  uint16_t Parameter  = rdm->Parameter;     
  bool8 handled = false;

  if (CmdClass == E120_SET_COMMAND) {
      *nackReason = E120_NR_UNSUPPORTED_COMMAND_CLASS;
    }
  
  return handled;
} `

peternewman, 2021-05-25:
> Yes, I was able to fix the code so DMX out throws to DMX address 1 regardless of RDM address and only to the pre-programmed size of the personality I set up. It also has a built-in function for identify. Here is the code!

Ah great!

Can I request/suggest you open a Pull Request against this repo to add it as another example (as personally I think it's that useful). It also makes it much easier to comment and review the code if required.

> I'm wondering how to remove all the unused options in the RDM menu of the device, so I only show the basics and not the special personalities the unit comes pre-programmed with.

Do you mean options that DmxSerial2 is offering which you don't feel are required? It looks to me like PID wise it should only offer some basic mandatory and useful ones by default.

I think productising the UID/manufacturer ID might be the only thing that needs sorting out, as I mentioned in https://github.com/mathertel/DmxSerial2/issues/27 .

Kevitto, 2021-05-25:
Created a Pull Request and added it as an example.  It's fairly clean, but again, I haven't been in the Arduino game very long, but this is a project I've been working on for a little while to make some of our dummy devices RDM compliant.  I use MAX485 chips in/out on Arduino Pro Minis.

peternewman, 2021-05-25:
Awesome thanks @Kevitto . You didn't clarify what you meant by this though:

> I'm wondering how to remove all the unused options in the RDM menu of the device, so I only show the basics and not the special personalities the unit comes pre-programmed with.

Kevitto, 2021-05-28:
Sorry, yes.  I meant, in the RDM controller, when I view the device, it has a long list of options available, but I don't need most if not all of them, all we need is a sensor read, address change and reset, but I'm not versed enough to know where to remove some of those options in DMXSerial2's source code.

peternewman, 2021-06-03:
Sorry @Kevitto can you share some screenshots/photos or just type out what option names you're seeing that you don't think you need.

It would also be good if you can confirm the hardware/software you're using for RDM control.