mathertel/DmxSerial2 #43: My RDM Arduino Uno device not working

zorarkdouble8, 2022-02-21:
So I'm using the RDMSerialRecv example (Link: https://github.com/mathertel/DmxSerial2/tree/master/examples/RDMSerialRecv) and it's not working. I am using the Chamsys Magicq software connected to a DMXKing ultraDMX2 Pro (Link: https://www.dmxking.com/usbdmx/ultradmx2-pro) which connects to my MAX485 RS485 transceiver IC chip which finally connects to my arduino Uno. 

The wiring looks like so:
[Wiring][1]

What happens is when I try to discover all my RDM devices, my arduino doesn't blink and my software doesn't detect anything. I can control my arduino through dmx using universe test but I can't detect rdm. Also, I'm using enttec Pro/Lumen Radio as a out type and I've messed with all the in types. 

  [1]: https://i.stack.imgur.com/qiYdS.png

zorarkdouble8, 2022-03-18:
So apparently my ultradmx2 pro won't work (finally got with chamsys support after coping the whole rdm mathertel circuit on a breadboard)

peternewman, 2022-03-18:
> The wiring looks like so: [Wiring](https://i.stack.imgur.com/qiYdS.png)

That looks about right, although I haven't checked the datasheets etc...

> What happens is when I try to discover all my RDM devices, my arduino doesn't blink and my software doesn't detect anything. I can control my arduino through dmx using universe test but I can't detect rdm. Also, I'm using enttec Pro/Lumen Radio as a out type and I've messed with all the in types.

I don't think you'd need an in type on Chamsys.

I'm biased, but you could also try the ultraDMX2 PRO with OLA ( https://www.openlighting.org/ola/ ) which will do RDM via the USB link natively and give you lots of useful debugging as to what's going on.

> So apparently magicq only uses artnet for rdm devices. Which means I have to somehow combine this library with a artnet library for an arduino. However, the ultradmx2 pro should've handled this.... Guess I'll have to do some more digging!

I was going to say, it looks like the ultraDMX2 PRO will do Art-Net RDM to DMX/RDM which should find the device fine, but would add another complication to testing it.

zorarkdouble8, 2022-03-26:
I got OLA to work on my raspberry pi 400 and it works with DMX. However, I still can't get RDM to work! It could be that I configured OLA wrong, the ultraDMX2 PRO, or the arduino configuration. Here's a screenshot of how I configured OLA: [Screenshot][1]

[1]: https://i.stack.imgur.com/py0s4.jpg

peternewman, 2022-03-28:
That looks about right @zorarkdouble8 .

You could try an `ola_rdm_set -u 1 --uid ffff:ffffffff identify_device 1` which should identify ALL devices so it should work even if you've got the wrong address or discovery isn't working.

Also olad -l 4 logs will help with diagnostics:
https://www.openlighting.org/ola/get-help/ola-faq/#How_do_I_get_olad_-l_4_logs

Especially if you do an `ola_rdm_discover -f -u 1`

zorarkdouble8, 2022-04-09:
I did the command: ola_rdm_set -u 1 --uid ffff:ffffffff identify_device 1 and it outputted: failed to send request. 

Another thing is I did the logs and the ola_rdm_discover command and it just says "Full RDM Discovery triggered for universe 1"

The only thing that might be an error is in the logs is: "common/io/Serial.cpp:170: Device /dev/ttyUSB0 locked by PID 1478 and process exists, failed to acquire lock"

I also tried to use ola_rdm_tests and:
[image]