mathertel/DmxSerial2 #46: RDM Buffer too small - truncated received messages and fails OLA rdm-tests

riwalker, 2023-01-31:
I notice that the receive buffer is 60 BYTES ?
  _// the byte array used while receiving and sending.
  byte buffer[60];_
RDM allows 231 Bytes, so why set at 60 ?

I noticed this as the OLA rdm-tests fail several tests, like E120_DEVICE_LABEL, > 32char
when set to 60, it truncates the RDM message at exactly 32char (I don't know why a buffer of 60 truncates the RDM Data at 32bytes)?
when i increase the buffer to [231] - we now capture the NACK for > 32 char test

BUT, not that the buffer is [231] it fails 41 instead of 6 !!!
any ideas ?

peternewman, 2023-02-01:
> I notice that the receive buffer is 60 BYTES ? _// the byte array used while receiving and sending. byte buffer[60];_ RDM allows 231 Bytes, so why set at 60 ?

Hi @riwalker 

This was fixed here:
https://github.com/mathertel/DmxSerial2/blame/98cdfc932de94a096e63033c18560dc69eb9f12c/src/DMXSerial2.cpp#L266-L270

And previously here:
https://github.com/mathertel/DmxSerial2/blame/98cdfc932de94a096e63033c18560dc69eb9f12c/src/DMXSerial2.h#L98

Specifically in https://github.com/mathertel/DmxSerial2/commit/e1f293e38fa5d1c28a467206e8517b7917feba1d

You'll need to either use Git or ask @mathertel nicely to release a new version.

> I noticed this as the OLA rdm-tests fail several tests, like E120_DEVICE_LABEL, > 32char when set to 60, it truncates the RDM message at exactly 32char (I don't know why a buffer of 60 truncates the [...]

The string should always be truncated by RDM at 32 bytes the other test checks for different responses when nacking a message that's too long but fits in the buffer from one that doesn't.

> BUT, not that the buffer is [231] it fails 41 instead of 6 !!! any ideas ?

What do you mean by 41 instead of 6? Tests? If so can you list which ones, ideally with debug output. It's probably just more slow EEPROM turnaround time.

riwalker, 2023-02-03:
Peter

 

Fixed the issue, by changing to Mega from Leonardo

Seems that by changing from 50 to 512, the RAM reduced too much and as soon as it went > 50-80%, even though no compiler issues, I started to get random ‘errors’

I thought it was my code with additional serial and watchdog, but stripping all that out and using the GOT code, and simply adding a bunch of Serial.println to add ‘variables’ into RAM, it started to show the same symptom (no compiler errors, just runtime errors), So its either Leonardo specifically, or just resources getting low (with the [50] to [231] byte increase)

Changing to the larger Mega, I see no errors whatsoever.

And now passing with just 4 OLA ‘Fails’ and plan to clear them down…

 

p.s still don’t have OLA rdm-tests running on Bullseye – would Really appreciate all the steps to download and build an image, way too many libraries missing and make errors for me to debug

 

Rob

 

https://github.com/mathertel/DmxSerial2/blame/98cdfc932de94a096e63033c18560dc69eb9f12c/src/DMXSerial2.cpp#L266-L270

https://github.com/mathertel/DmxSerial2/blame/98cdfc932de94a096e63033c18560dc69eb9f12c/src/DMXSerial2.h#L98

Specifically in  <https://github.com/mathertel/DmxSerial2/commit/e1f293e38fa5d1c28a467206e8517b7917feba1d> e1f293e