mathertel/DmxSerial2 #51: I2C PROBLEM

curta89, 2024-01-19:
WHY DO I HAVE PROBLEMS WHEN THE DISPLAY IS INITIALIZED?
SERIAL COMMUNICATION OR DMX COMMUNICATION STOPS WORKING:

#include <Wire.h>
#include <LiquidCrystal_I2C.h>
#include <DMXSerial2.h>

// ... (rest code)

// Set the I2C address of your LCD
#define I2C_ADDR    0x27

// Set the LCD dimensions
#define LCD_COLS    16
#define LCD_ROWS    2

// Create an instance of the LiquidCrystal_I2C class
LiquidCrystal_I2C lcd(I2C_ADDR, LCD_COLS, LCD_ROWS);

// ... (rest code)

void setup() {
  // ... (rest code)

  // Initialize the LCD
  lcd.begin(LCD_COLS, LCD_ROWS);

#if defined(SERIAL_DEBUG)
  // ... (rest code)
#endif
}

void loop() {
  // ... (rest code)

  // Display DMX address on the LCD
  lcd.clear();
  lcd.print("DMX Address:");
  lcd.setCursor(0, 1);
  lcd.print(DMXSerial2.getStartAddress());

  // ... (rest code)

mathertel, 2024-01-19:
The DMXSerial2 library completely runs in the background by using interrupts.

The effect you describe is likely an effect of another library (i2c?) that stops interrupts,
You did not give a glue on how i2c is implemented on the device you use - maybe it is a software based implementation that
stops all interrupts while transferring data ?

closed 2024-01-19 by mathertel

curta89, 2024-01-20:
After some tests, also using your i2c library, I discovered that in reality the dmx communication works perfectly.
There are problems when the master device searches for the slave device (receiver) via rdm.
He is found after a long time.
Once found functions like "change address" work.
This only happens if in the loop I write on the display. However, keeping the LCD initialized.

Do you have advice? Could it be a memory problem?

curta89, 2024-01-20:
ok, I found that a 5ms delay in the loop (the one it probably takes for writing to the display) does not allow the master to recognize the slave via rdm...